tzapu/WiFiManager #1718: error Exception (3) on ESP8266 with version 2.0.17 and not 2.0.16

PhDSM, 2024-03-10:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:	ESP8266 D1 Mini Board 		

Core Version:      	2.4.0, staging

### Description

error Exception (3)  occurs when calling AP webpage,  even with code example basic.ino on version 2.0.17.
However it works ok with version 2.0.16-rc 2 

### Settings in IDE

Module:  		Generic ESP8266 Module

Additional libraries: 

### Sketch
see  basic.ino  in your examples folder
```

### Debug Messages

11:10:14.771 -> *wm:resetSettings 
11:10:15.377 -> *wm:SETTINGS ERASED 
11:10:15.377 -> *wm:AutoConnect 
11:10:15.377 -> *wm:No wifi saved, skipping 
11:10:15.377 -> *wm:AutoConnect: FAILED for  2 ms
11:10:15.377 -> *wm:StartAP with SSID:  ESPWIFIMNGR
11:10:16.410 -> *wm:AP IP address: 192.168.4.1
11:10:16.410 -> *wm:Starting Web Portal 
11:10:48.824 -> 
11:10:48.824 -> Exception (3):
11:10:48.824 -> epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402485fd depc=0x00000000
11:10:48.859 -> 
11:10:48.859 -> >>>stack>>>
11:10:48.859 -> 
11:10:48.859 -> ctx: cont
11:10:48.859 -> sp: 3ffff930 end: 3fffffc0 offset: 01a0
11:10:48.859 -> 3ffffad0:  4024856b 3ffffb20 3ffffd60 40201878  
11:10:48.859 -> 3ffffae0:  6f666e00 3ffffe00 80fffb5c 4020e656  
11:10:48.859 -> 3ffffaf0:  3ffefd59 00000000 80fffb50 4020e480  
11:10:48.860 -> 3ffffb00:  3ffffdf0 00000148 3ffffd60 40203cec  
11:10:48.860 -> 3ffffb10:  00000000 3ffffe60 3ffffd60 40209010  
11:10:48.860 -> 3ffffb20:  3fff04cc 0098009f 00000020 401009cb  
11:10:48.889 -> 3ffffb30:  3fff0824 3ffef958 402014a8 00000001  
11:10:48.889 -> 3ffffb40:  3ffef9e4 3ffef958 3ffffbdc 4020dee9  
11:10:48.889 -> 3ffffb50:  3fff03ac 0028002f 00fef934 3fff084c  
11:10:48.889 -> 3ffffb60:  0e350e3f 00000001 3ffffbb0 00000001  
11:10:48.889 -> 3ffffb70:  3ffffbb0 3ffefa44 00000000 40203cec  
11:10:48.889 -> 3ffffb80:  00000001 00000001 3ffefa1c 40210b27  
11:10:48.889 -> 3ffffb90:  00000001 40201148 3ffefa1c 401000e1  
11:10:48.921 -> 3ffffba0:  3ffefa1c 3ffef974 3ffefa1c 40203d22  
11:10:48.921 -> 3ffffbb0:  3ffe0000 80ffc6fc 80000000 3ffffd60  
11:10:48.921 -> 3ffffbc0:  3ffefa1c 3ffef974 3ffef934 4020928a  
11:10:48.921 -> 3ffffbd0:  0054002f 00000000 81c6a7f0 3fff0800  
11:10:48.921 -> 3ffffbe0:  0000007f 80005054 40100200 00008562  
11:10:48.921 -> 3ffffbf0:  00000001 3ffef974 d9999999 00000000  
11:10:48.921 -> 3ffffc00:  00000001 00000000 4bc6a7f0 00000000  
11:10:48.953 -> 3ffffc10:  00000001 3ffef958 40100290 00000000  
11:10:48.953 -> 3ffffc20:  00000001 3ffef958 3ffef934 4020939f  
11:10:48.953 -> 3ffffc30:  009a3a4b 00000000 3fff011c 4020c34b  
11:10:48.953 -> 3ffffc40:  3fff00e4 3ffffe60 3ffffd60 4020cad0  
11:10:48.953 -> 3ffffc50:  00000000 3ffffe60 3ffffd60 402095c5  
11:10:48.953 -> 3ffffc60:  00000000 00000000 00000001 40100198  
11:10:48.953 -> 3ffffc70:  40210000 0104a8c0 3ffffd60 00000000  
11:10:48.985 -> 3ffffc80:  00000000 3ffffe60 3ffffd60 00000000  
11:10:48.985 -> 3ffffc90:  00000000 3ffffe60 3ffffd60 402099dd  
11:10:48.985 -> 3ffffca0:  3ffffd00 3ffffd00 3ffffce0 4020e2b0  
11:10:48.985 -> 3ffffcb0:  3ffffe60 3ffffd60 3ffffce0 4020dee9  
11:10:48.985 -> 3ffffcc0:  3ffffd00 3ffffd0c 4024c090 4020df08  
11:10:48.985 -> 3ffffcd0:  3ffe861c 3ffffd60 3ffffd18 4020dee9  
11:10:48.985 -> 3ffffce0:  736d2000 4020aa00 3ffffd60 3ffe8747  
11:10:49.017 -> 3ffffcf0:  00000001 3ffee9b8 3ffffd60 40209baf  
11:10:49.017 -> 3ffffd00:  736d2000 0002c700 80feebb0 736d2000  
11:10:49.017 -> 3ffffd10:  3ffffe00 80feebb0 00000000 3ffeeb50  
11:10:49.017 -> 3ffffd20:  80ffdad0 3ffeebb0 000001f4 3ffeeb50  
11:10:49.017 -> 3ffffd30:  3ffe861c 00000000 00000001 40202dd2  
11:10:49.017 -> 3ffffd40:  3fffdad0 00000001 00000002 3ffeeb50  
11:10:49.017 -> 3ffffd50:  3fffdad0 00000000 3ffeea20 40201086  
11:10:49.017 -> 3ffffd60:  3fff00e4 3ffef934 3ffefd54 3ffefd59

tablatronix, 2024-03-10:
Will test, have you tried a full erase?

tablatronix, 2024-03-10:
No issue here

```
*wm:[2] v2.0.17  D:4
*wm:[2] [SYS] getCoreVersion():          3.1.2
*wm:[2] [SYS] system_get_sdk_version():  2.2.2-dev(38a443e)
*wm:[2] [SYS] system_get_boot_version(): 31
*wm:[2] [SYS] getFreeHeap():             47520
```

PhDSM, 2024-03-22:
Here is info about  the ESP I used 
I have several of then and the problem occurs on all of them

esp8266
________________________________________
Chip ID c88a97
Flash chip ID 1450117
Flash size 4194304 bytes
Real flash size 4194304 bytes
Core version 2_6_3
Boot version 31
CPU frequency 80MHz
Memory - Free heap 42448 bytes available
Memory - Sketch size
Used / Total bytes
344032 / 962528

GodisGod, 2024-04-13:
same question

real-bombinho, 2024-04-28:
I hit the Exception (3) as well on some older NodeMCU1.0 boards but not on the ESP01 and D1 mini.

It seems to be raised due to no established connection on a first glance. 
Albeit the NodeMCU projects are some legacy projects, nursed along with little remaining memory. 

ESP01 and D1 mini use lesser memory and have less web dependent functionality.

jay6621, 2024-05-26:
I got the same issue, by using me-no-dev's esp exception decoder I got below result 
```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4000bf64
EXCVADDR: 0x4026b2e5

Decoding stack results
0x402138b3: String::String(char const*) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 34
0x402091e8: WiFiManager::getMenuOut() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/WString.h line 191
0x40213e1a: String::concat(__FlashStringHelper const*) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 395
0x40213c44: String::String(__FlashStringHelper const*) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 46
0x4020b770: esp8266webserver::FunctionRequestHandler ::handle(esp8266webserver::ESP8266WebServerTemplate &, HTTPMethod, String) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 42
0x402108c8: WiFiManager::handleRoot() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/WString.h line 124
0x401009c3: free(void*) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x40208ce8: std::_Function_base::_Base_manager   >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\aj\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x402136b5: String::invalidate() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 140
0x40213fa4: String::equals(String const&) const at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 502
0x4020b770: esp8266webserver::FunctionRequestHandler ::handle(esp8266webserver::ESP8266WebServerTemplate &, HTTPMethod, String) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 42
0x4021e6f7: std::_Function_handler    (WiFiManager*)> >::_M_invoke(std::_Any_data const&) at c:\users\aj\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x401000e1: std::function ::operator()() const at c:\users\aj\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2465
0x4020b7a8: esp8266webserver::FunctionRequestHandler ::handle(esp8266webserver::ESP8266WebServerTemplate &, HTTPMethod, String) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 49
0x40210b42: esp8266webserver::ESP8266WebServerTemplate ::_handleRequest() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 681
0x40100290: millis() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40210c53: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 340
0x402077ec: WiFiUDP::parsePacket() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiUdp.cpp line 199
0x4021e03c: DNSServer::processNextRequest() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\DNSServer\src\DNSServer.cpp line 166
0x40210e79: WiFiManager::processConfigPortal() at C:\Users\AJ\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 875
0x40100198: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40211291: WiFiManager::startConfigPortal(char const*, char const*) at C:\Users\AJ\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 807
0x40213a74: String::operator=(String const&) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 262
0x402136b5: String::invalidate() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 140
0x402136d4: String::~String() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 125
0x402136b5: String::invalidate() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 140
0x40204200: ESP8266WiFiSTAClass::hostname(char const*) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 587
0x40211463: WiFiManager::autoConnect(char const*, char const*) at C:\Users\AJ\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 388
0x4020a76c: WiFiManager::addParameter(WiFiManagerParameter*) at C:\Users\AJ\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 184
[... cut ...]
0x40212710: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char, bool) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\HardwareSerial.cpp line 51
0x40201669: setup() at D:\f3\superf3/superf3.ino line 114
0x40214ab4: loop_wrapper() at C:\Users\AJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 194
```

Board : D1 Mini (says ESP8266MOD)
Error : same as OP - error Exception (3) occurs when calling AP webpage

jay6621, 2024-05-26:
with board manager url `http://arduino.esp8266.com/stable/package_esp8266com_index.json`
and esp 8266 board version `3.1.2` solves issue for me
leaving with another library's issue.