tzapu/WiFiManager #1721: If anyone wants to change the name (SSID) of the Access Point (AP) that appears (with the IP address 192.168.4.1)

rtek1000, 2024-03-15:
If anyone wants to change the name (SSID) of the Access Point (AP) that appears (with the IP address 192.168.4.1), go to line 685. There seems to be a lack of alternatives in this part.

```C++
boolean WiFiManager::startConfigPortal() {
  String ssid = getDefaultAPName(); // <----------- SSID here: ESP32_XXXX
  return startConfigPortal(ssid.c_str(), NULL);
}
```

Example:
```C++
boolean WiFiManager::startConfigPortal() {
  String ssid = "My_AP"; //getDefaultAPName(); // <----------- SSID here: My_AP
  return startConfigPortal(ssid.c_str(), NULL);
}
```
https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.cpp#L686

----

On line 538 something about HostName seems to be implemented, but I don't know if this should also appear when you press the button, for on-demand mode.

```C++
  // set ap hostname
  #ifdef ESP32
    if(ret && _hostname != ""){
      bool res =  WiFi.softAPsetHostname(_hostname.c_str());
      #ifdef WM_DEBUG_LEVEL
      DEBUG_WM(WM_DEBUG_VERBOSE,F("setting softAP Hostname:"),_hostname);
      if(!res)DEBUG_WM(WM_DEBUG_ERROR,F("[ERROR] hostname: AP set failed!"));
      DEBUG_WM(WM_DEBUG_DEV,F("hostname: AP: "),WiFi.softAPgetHostname());
      #endif
   }
```

https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.cpp#L538

tablatronix, 2024-03-15:
Its the first argument in the function.. what do you mean?

startConfigPortal(ssid,..

rtek1000, 2024-03-15:
The initial post about the 686 line was with ESP32.

It also occurs with the ESP8266, in the line [388](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.cpp#L388)

```C++
  // not connected start configportal
  bool res = startConfigPortal(apName, apPassword);
```

At the beginning of Main there is wm.setHostname("MDNSEXAMPLE"), [line 37, onDemandNonBlocking.ino](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L37).

Shouldn't I have used this same "MDNSEXAMPLE" as the SSID?

rtek1000, 2024-03-15:
It appears there is a way to [read the SSID](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.cpp#L3226), but there is no way to Set it

```C++
/**
 * get config portal AP SSID
 * @since 0.0.1
 * @access public
 * @return String the configportal ap name
 */
String WiFiManager::getConfigPortalSSID() {
  return _apName;
}
```

rtek1000, 2024-03-15:
I managed to solve it, I was missing using:

[autoConnect](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.h#L257C5-L257C81)(apName, apPassword)

[startConfigPortal](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.h#L261C1-L261C13)(apName, apPassword)

```C++
char const * portal_SSID = {"DEVICE1"};
char const * portal_PASSWORD = {"12345678"}; // Need >= 8 chars

wm.autoConnect(portal_SSID, portal_PASSWORD);
wm.startConfigPortal(portal_SSID, portal_PASSWORD);
```

wm.[autoConnect](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L40)(portal_SSID, portal_PASSWORD);

wm.[startConfigPortal](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L74)(portal_SSID, portal_PASSWORD);

Maybe it would be easier if there was a function to set this, instead of the user declaring the SSID and password on each line they use when the portal starts.

Thank you.

rtek1000, 2024-03-15:
Should the [onDemandNonBlocking.ino](https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino) example always be non-blocking?

If the ESP is not connected, how long will the portal wait to go to the loop() routine?

When I comment out the wm.autoConnect() [line](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L40), then the led blinks, but if I leave the line, the portal is activated and remains active indefinitely.

I used wm.[setConfigPortalTimeout](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.h#L333)(timeout), but it still takes the timeout time to go to loop().

rtek1000, 2024-03-15:
> Should the [onDemandNonBlocking.ino](https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino) example always be non-blocking?
>
> If the ESP is [...]

The [line](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L39) needed to be uncommented:

`// wm.setConfigPortalBlocking(false);`

rtek1000, 2024-03-15:
> The [line](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L39) needed to be uncommented:
>
> `// [...]

If I use the portal Timeout, how can I know if the portal has already been stopped?

Note: On line [599](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.cpp#L599), a form of time control that is not recommended appears to be being implemented, in the case of recycling the variable returned to millis().

```C++
    // handle timed out
    if(millis() > _configPortalStart + _configPortalTimeout){
      #ifdef WM_DEBUG_LEVEL
      DEBUG_WM(F("config portal has timed out"));
      #endif
      return true; // timeout bail, else do debug logging
    } 
```

But on line [607](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.cpp#L607), A recommended form of time control is being implemented

```C++
      // log timeout time remaining every 30s
      if((millis() - timer) > logintvl){
        timer = millis();
        #ifdef WM_DEBUG_LEVEL
        DEBUG_WM(WM_DEBUG_VERBOSE,F("Portal Timeout In"),(String)((_configPortalStart + _configPortalTimeout-millis())/1000) + (String)F(" seconds"));
        #endif
      }
```

rtek1000, 2024-03-15:
> If I use the portal Timeout, how can I know if the portal has already been stopped?

I found this: [getConfigPortalActive()](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/WiFiManager.h#L489)

For the portal to deactivate (timeout), I changed the initial  value if [portalRunning](https://github.com/tzapu/WiFiManager/blob/d82d0a1b9fca741b9ec44accdf553606a6576dda/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino#L23) to true

My sketch now (ESP8266):
- It works like a WiFi router, the user have to press the button for 15 seconds
-  - The settings are reset.
-  - Wait for user to release button.
-  - LED goes out (on NodeMCU off: HIGH)
- The portal is only active for the timeout period, to avoid unnecessary exposure of the portal.
- The LED flashes differently when the portal is active

Sketch:
```C++
#include <Arduino.h>

/**
 * OnDemandNonBlocking.ino
 * example of running the webportal or configportal manually and non blocking
 * trigger pin will start a webportal for 120 seconds then turn it off.
 * startAP = true will start both the configportal AP and webportal
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// include MDNS
#ifdef ESP8266
#include <ESP8266mDNS.h>
#elif defined(ESP32)
#include <ESPmDNS.h>
#endif

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0 // ESP8266 GPIO0 (NodeMCU D3)

#define LED_BUILTIN_D4 2 // ESP8266 GPIO2 (NodeMCU D4)

char const *portal_SSID = {"DEVICE1"};
char const *portal_PASSWORD = {"12345678"}; // Need >= 8 chars

WiFiManager wm;

unsigned int timeout = 120; // seconds to run for
unsigned int startTime = millis();
//bool portalRunning = false;
bool startAP = true; // start AP and webserver if true, else start only webserver

uint32_t millis1 = 0;
uint32_t millis2 = 0;

void doWiFiManager();

void setup()
{
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run once
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(1000);
  Serial.println("\n Starting");

  pinMode(TRIGGER_PIN, INPUT_PULLUP);
  pinMode(LED_BUILTIN_D4, OUTPUT);
  digitalWrite(LED_BUILTIN_D4, HIGH);

  wm.setConfigPortalTimeout(timeout);

  // wm.resetSettings();
  wm.setHostname("MDNSEXAMPLE");
  // wm.setEnableConfigPortal(false);
  wm.setConfigPortalBlocking(false);
  // wm.autoConnect();

  wm.autoConnect(portal_SSID, portal_PASSWORD);
}

void loop()
{
#ifdef ESP8266
  MDNS.update();
#endif
  doWiFiManager();
  // put your main code here, to run repeatedly:

  if (wm.getConfigPortalActive() == true)
  {
    if ((millis() - millis2) > 500)
    {
      millis2 = millis();

      digitalWrite(LED_BUILTIN_D4, !digitalRead(LED_BUILTIN_D4));
    }
  } else {
    if ((millis() - millis2) > 5000)
    {
      millis2 = millis();
    }

    if (((millis() - millis2) < 500))
    {
      digitalWrite(LED_BUILTIN_D4, LOW);
    } else if (((millis() - millis2) > 500) && ((millis() - millis2) < 1000))
    {
      digitalWrite(LED_BUILTIN_D4, HIGH);
    }
  }
}

void doWiFiManager()
{
  // is auto timeout portal running
  if (wm.getConfigPortalActive()) //portalRunning
  {
    wm.process(); // do processing

    // check for timeout
    if ((millis() - startTime) > (timeout * 1000))
    {
      Serial.println("portaltimeout");
      //portalRunning = false;
      if (startAP)
      {
        wm.stopConfigPortal();
      }
      else
      {
        wm.stopWebPortal();
      }
    }
  }

  // is configuration portal requested?
  if (digitalRead(TRIGGER_PIN) == LOW && (!wm.getConfigPortalActive())) // !portalRunning
  {
    // Minimum button press time to activate the portal
    if ((millis() - millis1) > 15000)
    {
      digitalWrite(LED_BUILTIN_D4, HIGH);

      // Wait for user to release button
      while(digitalRead(TRIGGER_PIN) == LOW) {
        yield();
      }

      if (startAP)
      {
        Serial.println("Button Pressed, Starting Config Portal");
        wm.resetSettings();
        wm.setConfigPortalBlocking(false);
        wm.startConfigPortal(portal_SSID, portal_PASSWORD);
      }
      else
      {
        Serial.println("Button Pressed, Starting Web Portal");
        wm.startWebPortal();
      }
      //portalRunning = true;
      startTime = millis();
    }
    else
    {
      digitalWrite(LED_BUILTIN_D4, LOW);
    }
  }
  else
  {
    millis1 = millis();
  }
}
```

(Now I think I can go to the[ auto update part via email](https://github.com/mobizt/ESP-Mail-Client/blob/5c6e231e181ff863a325bba38eb692e88c087abb/examples/IMAP/Firmware_Update_Attachment/Firmware_Update_Attachment.ino)

Thank you.

tablatronix, 2024-03-16:
The ondemand part is non blocking but not autoconnect is not in that example, // wm.setConfigPortalBlocking(false);

I see the millis rollover issue, thanks

Looks like you figured the rest out. I suggest you read the .h file the super example and the wiki https://github.com/tzapu/WiFiManager/wiki/Methods

closed 2024-03-16 by rtek1000

rtek1000, 2024-03-18:
Hello, I would like to suggest that an example be made available (if possible) with a login to access the portal, it is the standard for almost all routers I have seen, I found a project that has this, perhaps it could serve as an example of how to implement it :

https://github.com/rtek1000/esp32-asyncwebserver-fileupload-example/tree/master/example-02

Login to access the portal:
[image]

Additional page that a commercial router shows if the user is unable to log in:
[image]

[info.zip](https://github.com/tzapu/WiFiManager/files/14630384/info.zip)

tablatronix, 2024-03-19:
#1669