tzapu/WiFiManager #1723: Wifimanager and ESP Mail Client

imoldovavan, 2024-03-19:
I get this error in this code:
`smtp.setClient(&wm);`
src\main.cpp:84:18: error: cannot convert 'WiFiManager*' to 'Client*'
   84 |   smtp.setClient(&wm);
      |                  ^~~
      |                  |
      |                  WiFiManager*

Can ESP Mail Client be used with wifimanager???
Thank you...

tablatronix, 2024-03-19:
Not like that, whats the syntax for that library what are you suppose to be passing to it ?

imoldovavan, 2024-03-20:
This is sample code from their examples/SMTP/External_Client/Generic/Generic.io

`// Your network client class object e.g. WiFiClient, EthernetClient and GSMClient`
` NetworkClient net_client;`

I replaced Network Client with WiFiManager  and net_client with wm so my code is:
`WiFiManager wm;`

The link to this example code is 
[[examples/SMTP/External_Client/Generic](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/External_Client/Generic/Generic.ino)](url)
Line 44 and 83

here is my code:

```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <ESP_Mail_Client.h>

#define SMTP_HOST "xxxxxxxxxxxxx"
#define SMTP_PORT esp_mail_smtp_port_587
#define AUTHOR_EMAIL "xxxxxxxxxxxxxm"
#define AUTHOR_PASSWORD "xxxxxxxxxxx"
#define RECIPIENT_EMAIL "xxxxxxxxxx"

// // put function declarations here:
// int myFunction(int, int);
int upone = 0;

SMTPSession smtp;

// Your network client class object e.g. WiFiClient, EthernetClient and GSMClient
WiFiManager wm;

void networkConnection()
{
  // Neywork connection code here
  //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  //  WiFiManager wm;

  // reset settings - wipe stored credentials for testing
  // these are stored by the esp library
  // wm.resetSettings();

  // Automatically connect using saved credentials,
  // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
  // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
  // then goes into a blocking loop awaiting configuration and will return success result

  bool res;
  // res = wm.autoConnect(); // auto generated AP name from chipid
  // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
  res = wm.autoConnect("AutoConnectAP"); // password protected ap

  if(!res) {
      Serial.println("Failed to connect");
      // ESP.restart();
  } 
  else {
      //if you get here you have connected to the WiFi    
      Serial.println("connected...yeey :)");
  }
}

// Define the callback function to handle server status acknowledgement
void networkStatusRequestCallback()
{
  // Set the network status based on your network client
  smtp.setNetworkStatus(true /* or false */);
}

void smtpCallback(SMTP_Status status);

void setup() {
  // WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  // it is a good practice to make sure your code sets wifi mode how you want it.

  // put your setup code here, to run once:
  Serial.begin(115200);

  networkConnection();

  MailClient.networkReconnect(true);

  smtp.debug(1);

  smtp.callback(smtpCallback);

  Session_Config config;

  config.server.host_name = SMTP_HOST;
  config.server.port = SMTP_PORT;
  config.login.email = AUTHOR_EMAIL;
  config.login.password = AUTHOR_PASSWORD;
  config.login.user_domain = F("127.0.0.1");

  config.time.ntp_server = F("pool.ntp.org,time.nist.gov");

  smtp.setClient(&wm);

  smtp.networkStatusRequestCallback(networkStatusRequestCallback);

  smtp.networkConnectionRequestCallback(networkConnection);

  if (!smtp.connect(&config))
  {
    MailClient.printf("Connection error, Status Code: %d, Error Code: %d, Reason: %s\n", smtp.statusCode(), smtp.errorCode(), smtp.errorReason().c_str());
    return;
  }

  if (!smtp.isLoggedIn())
  {
    Serial.println("Error, Not yet logged in.");
  }
  else
  {
    if (smtp.isAuthenticated())
      Serial.println("Successfully logged in.");
    else
      Serial.println("Connected with no Auth.");
  }

  SMTP_Message message;

  message.sender.name = F("ESP Mail");
  message.sender.email = AUTHOR_EMAIL;
  message.subject = F("Test sending plain text Email");
  message.addRecipient(F("Someone"), RECIPIENT_EMAIL);

  message.text.content = "This is simple plain text message";

  if (!MailClient.sendMail(&smtp, &message))
    MailClient.printf("Error, Status Code: %d, Error Code: %d, Reason: %s\n", smtp.statusCode(), smtp.errorCode(), smtp.errorReason().c_str());

  MailClient.printf("Free Heap: %d\n", MailClient.getFreeHeap());
    

}

void loop() {
  Serial.print("Loop waiting:");
  Serial.println(upone);
  delay(1000);
  upone++;
}

void smtpCallback(SMTP_Status status)
{
  Serial.println(status.info());

  if (status.success())
  {
    Serial.println("----------------");
    MailClient.printf("Message sent success: %d\n", status.completedCount());
    MailClient.printf("Message sent failed: %d\n", status.failedCount());
    Serial.println("----------------\n");

    for (size_t i = 0; i < smtp.sendingResult.size(); i++)
    {
      SMTP_Result result = smtp.sendingResult.getItem(i);

      MailClient.printf("Message No: %d\n", i + 1);
      MailClient.printf("Status: %s\n", result.completed ? "success" : "failed");
      MailClient.printf("Date/Time: %s\n", MailClient.Time.getDateTimeString(result.timestamp, "%B %d, %Y %H:%M:%S").c_str());
      MailClient.printf("Recipient: %s\n", result.recipients.c_str());
      MailClient.printf("Subject: %s\n", result.subject.c_str());
    }
    Serial.println("----------------\n");
    smtp.sendingResult.clear();
  }
}
``` 

Thank you for your help.

tablatronix, 2024-03-21:
Wm doesn't replace wifi client, you will have to use the mail constructor without a wifi connection and use wm before hand. Or use the regular one and handle wm before that if

imoldovavan, 2024-03-22:
Got it. Thank you.

closed 2024-03-22 by imoldovavan