tzapu/WiFiManager #1725: Feature configurable? WiFi AP started when local WiFi is down

sany3001, 2024-03-22:
As soon as there are no Discussions enabled on this repo, I need to ask question here:

From project description:
* if this is unsuccessful (or no previous network saved) it moves the ESP into Access Point mode and spins up a DNS and WebServer (default ip 192.168.4.1)

I want to avoid in my project on many of my devices just because my WiFi is down that AP gets activated when there are WiFi credentials stored previously.

Can I achieve that be (re)configuring WiFiManager?

tablatronix, 2024-03-22:
Yeah I do not have privs to make those changes, no problem.

You can use only ondemand, or a reset mechanism, long press etc.
Add a configportaltimeout
Add a user code check for x many minutes or reboots before starting cp.

tablatronix, 2024-03-22:
```
    // if true (default) then start the config portal from autoConnect if connection failed
    void          setEnableConfigPortal(boolean enable);
```

closed 2024-03-22 by tablatronix