tzapu/WiFiManager #1734: Question: Is possible change deep link to redirect to an App instead of Captive Portal

JonyPower, 2024-04-16:
First of all thanks to the amazing work with this library!!!
I would like to know how could be possible chage the behavior after login in wifi, instead of redirect to Captive Portal I wanna open my app if it was already instaled or browser to my website if not.

ateker01, 2024-04-16:
If I understand this correctly +1 this...

When the captive portal launches, it redirects to http://192.168.4.1/
how can i change the redirect link to something else? for example .... http://192.168.4.1/wifi? instead

When the credentials are saved, the url directs to http://192.168.4.1/wifisave
how can i change the redirect link to something else? perhaps one URL if the credentials are correct, and another URL if the credentials are not correct

JonyPower, 2024-04-17:
"how can i change the redirect link to something else? for example .... http://192.168.4.1/wifi? instead"
Yeah could be something like this, customize the redirect address page that I could choose the link, like  "http://mywebsite.com/nameoftheproject/appstore"

The idea would be for the user to create an account on my App and the App sends the information via http post to be saved on the ESP, such as Firebase credentials or another backend.

tablatronix, 2024-04-20:
You can add your own routes check the SUPER example, not sure how you can override the root/ I have to think about this more but its not hard to poke in the code and change these