tzapu/WiFiManager #1735: Add new Release to platformio

dbambus, 2024-04-18:
Hey @tzapu, 

Thank you very much for your Release 2.0.17, it is absolutely brilliant and I already intend to use it in my repo for a word clock. However, for this project I use platformio and it would be super if the release could also be managed by their repo system. 

The previous release 0.16.0 is already there, but could you update it to the latest release as well? 

https://registry.platformio.org/libraries/tzapu/WiFiManager

Cheers David

FerFemNemBem, 2024-04-19:
Hi @dbambus,

what about adding:

lib_deps =
https://github.com/tzapu/WiFiManager.git#v2.0.17

to your "platformio.ini"?

Cheers, FFNB

tablatronix, 2024-04-20:
This should have updated automatically when I made the release..
hmm

tablatronix, 2024-04-20:
I think platformio remove auto-update legacy from registry, you have to manually publish updates now, which means I think you need a pio organization ties to the repo, its possible only @tzapu can do this.. 

Looking into it

tablatronix, 2024-12-02:
I will have to provide my own fork of wm to manage a platform io library