tzapu/WiFiManager #1739: unable to do OTA update for esp32 s3. update failed due to error in downloading version. unable to find the where the exact issue is?

citygreenstech2019, 2024-05-15:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

Chalta, 2024-09-06:
Duplicate of #1741?

(Which is also not related to WiFi Manager)

tablatronix, 2024-09-09:
I never got it working assumed it was partition related and needed special code to handle that