tzapu/WiFiManager #1744: Get stored SSID before connection is made, to connect to strongest of multiple identical SSIDs

richardhawthorn, 2024-06-18:
Hi there

I'm trying to get the SSID of the stored network before the first connection is made (I know I can use WiFi.SSID() once the network is connected, but I don't want to connect first)

Background: I have a network with multiple identical SSIDs (and passwords), and I want to connect to the strongest of those.  If I have the SSID I can scan for all networks (matching on SSID), check which is the strongest, and use it's BSSID to connect to that access point specifically, giving me the strongest connection.

Thanks!

Richard

richardhawthorn, 2024-06-21:
For those looking to do the same thing, here are the function you need to access the SSID and password before the connection is made:

ESP8266
```
String getStoredSSID(){
  struct station_config conf;
  wifi_station_get_config(&conf);
  return (char *) conf.ssid;
}
```

ESP32
```
String getStoredSSID(){
  wifi_config_t cachedConfig; 
  esp_err_t configResult = esp_wifi_get_config( (wifi_interface_t)ESP_IF_WIFI_STA, &cachedConfig );
      
  if( configResult == ESP_OK ){
    return (char *) cachedConfig.ap.ssid;
  }
}
```

With both of these, you can get the saved password by replacing 'ssid' with 'password'.

closed 2024-06-21 by richardhawthorn

tablatronix, 2024-06-22:
```c++
 // WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN); // wifi_scan_method_t scanMethod
  // WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL); // wifi_sort_method_t sortMethod - WIFI_CONNECT_AP_BY_SIGNAL,WIFI_CONNECT_AP_BY_SECURITY
```