tzapu/WiFiManager #1746: 界面设置为中文的解决办法

RorinL, 2024-06-28:
[前往我的站点](https://fongdan.com/article/detail/19)

WiFiManager.h AP配网在使用的时候需要改为显示中文,开始以为是使用 setCountry("CN") 来设置,但是怎么改都没变化,看了WiFiManager.cpp 和 WiFiManager.h 才知道原来这个函数作用是设置 WiFi AP 模块的国家代码,国家代码决定了模块可以使用的无线信道和频率范围(不同的国家和地区有不同的无线电频谱规定); 例如，在美国（国家代码为 "US"），WiFi 信道通常从 1 到 11，而在欧洲，信道可以从 1 到 13;也就是说setCountry函数是用于确保 WiFi 模块遵循当地的无线电频谱规定

个人感觉没什么用吧，哈哈;rorinliang076 at/ gmail不过我也设置了"CN".

### 本人开发环境
arduino ide2.3.2
[WiFiManager版本2.0.17](https://github.com/tzapu/WiFiManager)
windows 10专业版

### 汉化方法
1. 先下载文件 [WiFiManager汉化文件](https://www.lanzoub.com/itPSy22ycqkh
), 解压后得到 wm_consts_cn.h以及wm_strings_cn.h,strings_cn.h;将三个文件放到WiFiManager根目录下;
WiFiManager一般安装在Arduino工具->首选项->项目文件夹地址下的 libraries 文件夹中fongdan,com
2. 打开 WiFiManager.h,修改 WM_STRINGS_FILE,将 wm_strings_en.h 修改为 wm_strings_cn.h
```
// Include wm strings vars fongdan,com
// Pass in strings env override via WM_STRINGS_FILE
#ifndef WM_STRINGS_FILE
#define WM_STRINGS_FILE "wm_strings_en.h" // 修改为 wm_strings_cn.h
#endif
#include WM_STRINGS_FILE
```
这样就可以完成汉化了

### 其他

或者会疑问，为什么我在我的代码文件不直接像下面这样?这样不就能改了吗，又不会修改源文件.
```
#define WM_STRINGS_FILE "wm_strings_cn.h"
#include <WiFiManager.h>
```
但是修改不了，我也不知道为什么
fongdan.com

如果你觉得我的解决办法好，同时你想将其添加到该开源项目，那么感谢你！同时请你能注明来源

[前往我的站点](https://fongdan.com/article/detail/19)

tablatronix, 2024-06-28:
Can you pull request this ?

`#define WM_STRINGS_FILE "wm_strings_cn.h"` only works when defined in build enviroment, wont work from .ino sketch

RorinL, 2024-06-28:
I'm not really good with this pull request, so maybe I'll just leave it here. One more thing. Are you saying my method doesn't work?

wzsx150, 2024-09-16:
我已经提交了中文翻译的PR，欢迎大家一起优化中文翻译。
I have submitted a PR for the Chinese translation, and I welcome everyone to help optimize the Chinese translation.
https://github.com/tzapu/WiFiManager/pull/1764

YourMao, 2024-09-26:
可用

444136347, 2024-10-01:
@RorinL 在页面上，会有点问题。
（1）点击选择对应WiFi，无法自动填入WiFi名称
（2）show Passpwd有点问题

zofui, 2025-03-24:
> [@RorinL](https://github.com/RorinL) 在页面上，会有点问题。 （1）点击选择对应WiFi，无法自动填入WiFi名称 （2）show Passpwd有点问题

他里面的onlick的字符o输入错了，你删除他的改成使用英文onclick就好了