tzapu/WiFiManager #1751: OnDemand AP fails with unhandled exception

kungfupizza, 2024-07-15:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP32-s3

Core Version:      	3.0.0, staging

### Description

When running WM as an ESP-IDF task and putting the WM into OnDemandPortal when a button is pressed, gets me to an unhandled exception. Very much reproducable at my end.

### Settings in IDE

Module:  			ESP32S3 M1

Additional libraries: 

### ESP IDF task

```cpp
static void wtask(void *pvParameters)
{
	WiFi.mode(WIFI_STA);
	wm.setConfigPortalTimeout(30);
	wm.setHostname(PRJ_CONFIG_WIFI_MANAGER_HOSTNAME);
	wm.autoConnect();

	wm.setAPCallback(ap_start_cb);
	wm.setSaveConfigCallback(wifi_connected_cb);

	while (true)
	{
		wm.process();
		vTaskDelay(pdMS_TO_TICKS(100));
	}
}
```

### on button press callback
```cpp
static void wifiSetupModeCb(void)
{
  if (wm.getConfigPortalActive())
  {
	  wm.stopConfigPortal();
  }
  wm.setConfigPortalTimeout(30);
  wm.startConfigPortal();
}
```
`NOTE:` Instance of the class is defined as a static in the code
`
static WiFiManager wm;
`
### Debug Messages

```
I (7074) wifi:mode : sta (f4:12:fa:86:76:08) + softAP (f4:12:fa:86:76:09)
I (7074) wifi:Total power save buffer number: 16
I (7075) wifi:Init max length of beacon: 752/752
I (7078) wifi:Init max length of beacon: 752/752
I (7085) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

0x4203a411: WiFiManager::processConfigPortal() at <path>/WiFiManager.cpp:860
```

`<path>/WiFiManager.cpp:860` points to 
```
//HTTP handler
server->handleClient();
```
in the code

tablatronix, 2024-07-16:
You should probably feed wdt in that loop

kungfupizza, 2024-07-16:
I bet the watchdog is turned off for this test.