tzapu/WiFiManager #1753: Bare ESP-IDF Support

italocjs, 2024-07-20:
Does this library supports esp-idf alone? without Arduino behind? will it ever support?

I've been trying to compile for a while, i would rather not adding arduino to my system. 

```
[env:native]
lib_ldf_mode = deep+
test_framework = unity
platform = native
build_flags = 
    -D USE_NATIVE_ENVIRONMENT
    -D PLATFORM_NATIVE
    -Iinclude
    -Ilib/filesystem
; lib_deps = 
    ; unity

[env:esp32dev]
build_type = debug
test_framework = unity
platform = espressif32
board = simovatrack130
framework = espidf
board_build.partitions = boards/partitions.csv
build_flags = 
    -D USE_ESP32_ENVIRONMENT
    -D PLATFORM_ESP32
    -Iinclude

upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
    esp32_exception_decoder
    send_on_enter
```

tavdog, 2025-05-10:
Me too. esp-idf needs a good wifimanager

tablatronix, 2025-05-12:
I imagine we could remove all arduino functions.
At that point its probably better to rewrite from scratch as most of this is library is handling arduino specific quirks