tzapu/WiFiManager #1754: Changing wifi without having to turn on the config portal

VladimirVecera, 2024-07-29:
First of all ... wifimanager is an absolutely amazing tool and I thank you for it. Thank you very much.
I have a question. I configure the ESP32 through the config portal, the ESP32 successfully connects to wifi. In a month I want to change wifi accesses. At the same time, I have my own web administration on ESP32. Is it possible to change wifi accesses without having to activate the config portal, which works on 192.168.4.1 and is it necessary to have a wifi device at hand to set up wifi? I would like to be able to configure the wifi via the web interface on the already connected wifi. I hope you understand me.

F1p, 2024-07-29:
wifiManager.startWebPortal();

At the end of setup to run the web server when on WiFi
And in loop to handle requests and modification:

wifiManager.process();

You may also need this if saving fields to a file:
if (shouldSaveConfig) { saveConfig(); }  // Handles WiFiManager Settings Changes

tablatronix, 2024-07-29:
You can do it in regular esp code via begin()

VladimirVecera, 2024-07-30:
I'm a bit confused. Can you send the best code? Thank you very much in advance for your help.

F1p, 2024-07-31:
> I'm a bit confused. Can you send the best code? Thank you very much in advance for your help.

Edit of basic example:

`#include <WiFiManager.h>

void setup() {
  WiFiManager wm;
  bool res;
  res = wm.autoConnect("AutoConnectAP", "password");
  if (!res) {
    Serial.println("Failed to connect");
  } else {
    Serial.println("connected...yeey :)");
  }

  wm.startWebPortal();  // Post connection WiFi Manager Portal Start
}

void loop() {
  wm.process();  // Handle requests
}
`

VladimirVecera, 2024-08-12:
The problem is that the web administration of the project is already running on the web interface #include <WebServer.h>
And if I want to change the wifi configuration to the already successfully connected wifi, it doesn't work. Does it even work?

`  server.on("/wifiportal", []() {
    Serial.println("Zapínám Wifimanager config portal na /wifi");
    wm.startWebPortal();  // Spuštění webového portálu
  });
`

tablatronix, 2024-08-12:
I thought you meant in code.. you want to start the webportal you cant of you are already running another webserver

VladimirVecera, 2024-08-13:
`#include <WiFi.h>
#include <WiFiManager.h>  // Knihovna WiFiManager
#include <ESPmDNS.h>       // Knihovna pro mDNS

WiFiManager wm;

void setup() {
  Serial.begin(115200);

  // Spustí WiFiManager a připojí se k WiFi síti
  if (!wm.autoConnect("AutoConnectAP")) {
    Serial.println("Failed to connect and hit timeout");
    // Možná chcete restartovat zařízení nebo jinak reagovat
    ESP.restart();
    delay(1000);
  }

  Serial.println("Connected to WiFi!");

  // Nastavení mDNS
  if (!MDNS.begin("esp32")) {  // "esp32" je jméno mDNS, zařízení bude dostupné na "http://esp32.local"
    Serial.println("Error setting up MDNS responder!");
  } else {
    Serial.println("mDNS responder started");
  }

  // Přidání vlastního handleru pro "/mycustomurl"
  wm.server->on("/mycustomurl", HTTP_GET, []() {
    wm.server->send(200, "text/plain", "Hello, this is a custom URL!");
  });

  // Spuštění serveru
  wm.server->begin();

  Serial.println("HTTP server started");
}

void loop() {
  // Je nutné pravidelně volat loop WiFiManageru
  wm.process();
  
  // Musíte pravidelně volat MDNS.update()
  MDNS.update();
}

`