tzapu/WiFiManager #1760: ESP32 getting back to config State when my router is off for sometime

danielgonzalvis, 2024-08-20:
I am using Wifimanager library where i manually enter the credentials and if in case of power cuttoff my router turn off then when power comes back then esp32 will go back to its config mode why is that is this present in library

klasik16, 2025-01-03:
Have same situation. Does anyone has sugesstion on that?

klasik16, 2025-01-03:
@tzapu please comment on that. becouse this is whole point of autoconnect.

tablatronix, 2025-01-03:
not sure what you are asking but this has been answered here countless times

closed 2025-01-03 by tablatronix

klasik16, 2025-01-03:
@tablatronix Could the answer be linked there? Searcihing for this a while. also so similar open questions. [https://github.com/tzapu/WiFiManager/issues/1752](url)

tablatronix, 2025-01-03:
Sorry the search sucks now, cant find anything

you are looking for configPortalTimeout, this will timeout the cp and you can recheck for wifi or reboot etc.

You can also turn off configPortal in autoconnect if wifi is saved, and use some kind of timer in your code to start it.
You can also set the connecttimeout time