tzapu/WiFiManager #1765: On Demand Configuration Portal do not work

mylms, 2024-10-06:
ESP32-C3
tzapu/WiFiManager@^2.0.17

I would like to use **auto connect** and **manually start the ConfigPortal** if manual configuration (MQTT etc.) is required. BUT there is "Do not use BOTH" in "On Demand Configuration Portal".
How to allow the user to automatically connect to a known WiFi (after start the device, if setting not required), but at the same time manually launch the Configuration portal with a button, for example?

- If I use "autoConnect" in setup ESP32 connects to Wifi
- If at the same time I use somewhere in the code (button event) "startConfigPortal", wifi is created, but it is not possible to connect to it (even the address 192.168.4.1 does not work).

- If I use "startConfigPortal" instead of "autoConnect" in setup, ESP does not connect to wifi.

- If I use only "startConfigPortal" it is possible to connect to ConfigPortal, but after exit it's not connected to wifi

`
void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    WiFiManager wifiManager;
    wifiManager.startConfigPortal("OnDemandAP");
    Serial.println("connected...yeey :)");  //**<<< NO, IT DOESN'T**
  }
}
`

mylms, 2024-10-06:
Problem solved!
I use WebServer `WebServer.begin();` after `wifiMngr.autoConnect`.
Just deteled `WebServer.begin();` and everything works!

smeisner, 2025-02-18:
@mylms I am confused. I am trying to do the same (after ESP32 is config'd and connected via WifiManager autoconnect()) I want to use a button to trigger the AP to change the config. I cannot get this working at all.

How did you resolve this?

tablatronix, 2025-02-18:
What does the example do?

smeisner, 2025-02-18:
The problem I was encountering was the AP was being created (according to the debug output), but when I connected to the CP via the AP (at 192.168.4.1), I got no response at all in my browser. And nothing interesting in the log output, except a client was connected.

I was using a project created by someone else as a starting point. What I found after reviewing all the sample code and comments here was the WiFiManager code was indeed correct, but the config portal wasn't working. Finally after digging through much of the existing code, I found it; A web server for OTA was being created on port 80. If I commented that out, WifiManager config portal worked fine. So I am using a different port for the OTA page.

It was the comment from @mylms about removing the `WebServer.begin()` that made me think to review the code.