tzapu/WiFiManager #1771: No need to display static IP fields on WIFI connection page -Help

gpena208777, 2024-10-26:
Hello, I am trying to hide as many fields and options as I can for the end user. When the user clicks and enters their wifi credentials, I don't want the other IP address fields underneath to display. 
Any way to hide them?

tablatronix, 2024-10-26:
Thats the default, there is a showipfields setting

gpena208777, 2024-10-26:
> Thats the default, there is a showipfields setting

wifiManager.setShowStaticFields(false);    

Should this be false?

tablatronix, 2024-10-31:
Are you doing something with static ip in your code that would cause them to show? Cause then yes, what do the examples do?

gpena208777, 2024-11-03:
> Are you doing something with static ip in your code that would cause them to show? Cause then yes, what do the examples do?

My code calls for a static IP to be set could this be why ?

tablatronix, 2024-11-03:
Yeah if you use static in code it automatically shows in web.

you can add that setShowStaticFields(boolean alwaysShow); as falss