tzapu/WiFiManager #1781: Change default charset UTF-8

mylms, 2024-11-19:
Is it possible change default charset from UTF-8 to ISO-8859-2? I use this encodig in VS Code (because encoding chars for external display).
I would like to get some texts from WiFi portal to ESP32 and there is some problems with decoding letters á, č, ď, é, ě, í, ň, ó, ř, š, ť, ú, ů, ý, ž, Á, Č, Ď, É, Ě, Í, Ň, Ó, Ř, Š, Ť, Ú, Ů, Ý, Ž

ThankYou

tablatronix, 2024-11-20:
You mean the web browser? You can add a header

mylms, 2024-11-20:
Thank you for your advice. Yes, I mean in web browser. I try to add 
`wm.setCustomHeadElement("<meta charset='ISO-8859-2'>");`

But anfortunatelly no change. Diacritic chars not working.

Here is code of web.
`<html>
<body>

<!DOCTYPE html><html lang='en'><head>**<meta charset='UTF-8'>**<meta  name='viewport' content='width=device-width,initial-scale=1,user-scalable=no'/><title>WiFiManager</title><script>...some stuff...</style>**<meta charset='ISO-8859-2'>**</head><body class='invert'><div class='wrap'><h1>WiFiManager</h1><h3>BeerMeterCfg</h3><form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/> ...some other stuff...
</body>
</html>`

[image]

mylms, 2024-11-20:
Here is code of webpage. I can't paste it directly.
[image]

assanges, 2025-01-22:
> Thank you for your advice. Yes, I mean in web browser. I try to add `wm.setCustomHeadElement("<meta charset='ISO-8859-2'>");`
> 
> But anfortunatelly no change. Diacritic chars not working.
> 
> [image]

Are you modifying the [`wm_strings_en.h`](../blob/master/wm_strings_en.h) file? 

Not sure if this will work but it's worth a try since I don't have an ESP board beside me.

Try to modify Line 28 in the i18n file (i.e., [`wm_string_en.h:28`](../blob/master/wm_strings_en.h#L28) or any self-created i18n file),
the charset in the HTML file was a hardcoded string, adding a custom HTML Head element would not work as you think.

```
const char WM_LANGUAGE[] PROGMEM = "en-US"; // i18n lang code (you might want to change this to "cs-CZ")

const char HTTP_HEAD_START[]       PROGMEM = "<!DOCTYPE html>"
"<html lang='en'><head>"
"<meta name='format-detection' content='telephone=no'>"
"<meta charset='UTF-8'>" // MODIFY CHARSET HERE, i.e. 'ISO-8859-2'
"<meta  name='viewport' content='width=device-width,initial-scale=1,user-scalable=no'/>"
"<title>{v}</title>";

```