tzapu/WiFiManager #1787: 13 custom fields renders 4/10 times on ESP8266

F1p, 2024-12-24:
I have a reasonable sized project, running 13 custom fields and have an issue on the ESP8266 where 4/10 times the custom fields render blank on refresh (as per screenshots) but issue doesn't seem to impact the ESP32. Any idea's why this might happen?

[image]
[image]

Running the latest master branch, 

ESP8266 running
Core version
3.1.2
Boot version
31
CPU frequency
160MHz
Memory - Free heap
12912 bytes available
Memory - Sketch size
Used / Total bytes
489296 / 3143504

F1p, 2024-12-29:
Might be an issue resolved by adjusting MMU to 16KB Cache + 48K IRAM instead of balanced (32/32K)

tablatronix, 2024-12-30:
Its probably memory, the html is output as a string on the stack, If you are using alot of params this can be optimized by using chunked outputting and allocated buffers, I think there is an issue here to start looking into it..

#1319 
#518

F1p, 2024-12-30:
> Its probably memory, the html is output as a string on the stack, If you are using alot of params this can be optimized by using chunked outputting and allocated buffers, I think there is an issue here to start looking into it..
> 
> #1319 #518

Thanks, will try chunked response for now