tzapu/WiFiManager #1790: Question: Does this library use EEPROM for store networks info(name and password) and what is the coordinates if it is?

IvanSkorocodov, 2025-01-05:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2025-01-05:
no, esp library handled the storage for wifi creds, user is responsible for params

timr49, 2025-04-17:
Hi @IvanSkorocodov, WiFiManager does not explicitly store or retrieve the WiFi SSID and password. Rather, it relies on the native esp8266 and esp32 support for doing that.

Custom parameters are another matter.  They are explained in README.md under the heading **Custom Parameters**.  The user of the library is responsible for storing and retrieving their values.

So, WiFiManager does not itself do any storing of information.