tzapu/WiFiManager #1794: WiFiManager not working on ESP32-S3 (ESP32-S3-DevKitC-1 board)

mrWheel, 2025-01-23:
Environment:
   VSCode, PlatformIO, Arduino Framework

Board:
  ESP32-S3-DevKitC-1 (Espressif)

platformio.ini:
```
[platformio]
workspace_dir = .pio.nosync
default_envs = plainWiFi

[env]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.filesystem = littlefs
framework = arduino
monitor_speed = 115200

[env:plainWiFi]
lib_deps = 
	${env.lib_deps}
	https://github.com/tzapu/WiFiManager.git

build_flags = 
     -DCORE_DEBUG_LEVEL=5

monitor_filters = 
	esp32_exception_decoder
```

Program Code:
```
#include <Arduino.h>

#if defined(ESP8266)
    #include <ESP8266WiFi.h>
#else
    #include <WiFi.h>
#endif

#include <WiFiManager.h>

WiFiManager wm;

void setup() 
{
  Serial.begin(115200);
  delay(2000);
  Serial.println("Lets get started ...");
  //-- WiFiManager
  Serial.println("Initiate WiFiManager ..");
  //-- Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;
  //-- reset settings - for testing
  //wm.resetSettings();
  //-- sets timeout until configuration portal gets turned off
  //-- useful to make it all retry or go to sleep
  //-- in seconds
  wm.setDebugOutput(true);
  wm.setConfigPortalTimeout(180);
  
  //-- fetches ssid and pass and tries to connect
  //-- if it does not connect it starts an access point with the specified name
  //-- here  "AutoConnectAP"
  //-- and goes into a blocking loop awaiting configuration
  if(!wm.autoConnect("ESP32S3-AP")) 
  {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //-- reset and try again
    ESP.restart();
  } 

  //-- if we get here we have connected to the WiFi
  Serial.println("Connected to WiFi");
  delay(1000); // Give connection time to stabilize
}

void loop() 
{
    wm.process();
}
```
After flashing the sketch to the board, it is immediately noticeable that the board does not start automatically.

So after manually pressing [reset], this is the output:
```
[   268][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   299][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(44) txPin(43)
[   307][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[   316][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
Lets get started ...
Initiate WiFiManager ..
*wm:v2.0.17  D:2
*wm:AutoConnect 
[  2329][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
E (4469) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x110e)

[  2369][V][WiFiGeneric.cpp:341] _arduino_event_cb*w(): STA Started
:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  58 ms
[  2387][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[  2396][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
*wm:StartAP with SSID:  ESP32S3-AP
[  2402][E][WiFiAP.cpp:169] softAP(): set AP config failed
[  2410][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
[  2411][V][WiFiGeneric.cpp:344] _arduino_event_cb(): STA Stopped
o  2423][D][WiFiGeneric.cpp:1040] _eventC[a l l2b4a2c3k]([)V:] [AWridFuiiGneon eErviecn.tc:p p3: 3-9 6S]T A__aSrTdOuPin
_event_cb(): AP Stopped
[  2436][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 11 - AP_STOP
*wm:[ERROR] There was a problem starting the AP 
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal 
[  2910][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  2926][V][WebServer.cpp:88] WebServer(): WebServer::Webserver(port=80)
``` 
these errors occur:
`[  2402][E][WiFiAP.cpp:169] softAP(): set AP config failed`
`*wm:[ERROR] There was a problem starting the AP `
... and although it then says that a web portal is being started on port 80, it is not visible in the available APs list.

My question is: Am I the only person who has trobles getting the WiFiManager to work with an (Espressif) ESP32-S3 board?

mrWheel, 2025-01-25:
Ok, this line:
```
E (4469) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x110e)
```
Seems to be the problem.

In the partition table there needs to be a `nvs` partition of at least 0x5000 and a `phy_init` partition of (at least?) 0x1000 and than the WiFi manager works as expected!

closed 2025-01-25 by mrWheel

tablatronix, 2025-01-25:
I have not had this issue that board definition should have a default partition table

mrWheel, 2025-01-25:
I have my own partition table which was in errer.