tzapu/WiFiManager #1797: updating ESP32 arduino core to 3.1.0 seems to break wm.startConfigPortal()

GrantSando, 2025-02-01:
If I call  wm.startConfigPortal(ssid, pass) with arduino-esp32 3.0.7 or earlier it creates the portal.

If I call it with arduino-esp32 3.1.0 or later installed it causes a "Guru Meditation Error: Core 1 panic'ed." and restarts device.

code looks like this,

#include "Arduino.h"
#include <WiFiManager.h>
#include <string.h>
#include <Preferences.h>

Preferences preferences;
WiFiManager wm;               
char ssidString[50] = "";
char pwString[50] = "";
const int setupPin = 21; 

void setup() {
    Serial.begin(9600);
    delay(1500); 
    pinMode(setupPin, INPUT_PULLUP);
    getWiFiCredentials();
//    InitWiFi();
}

void loop() {
  // put your main code here, to run repeatedly:

}

void getWiFiCredentials(){

  //load ssid and password from preferences
  preferences.begin("credentials", false);
    String ssid = preferences.getString("ssid", ""); 
    ssid.toCharArray(ssidString, sizeof(ssidString));
    String pw = preferences.getString("password", "");
    pw.toCharArray(pwString, sizeof(pwString));
  preferences.end();
  
  WiFi.mode(WIFI_STA);

  //if ssid and password have not been setup then start portal for user input
  if (ssid == "" | digitalRead(setupPin) == 0)
  {
    if (!wm.startConfigPortal("MY_WIFI", "PASS1234"))
    {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.restart();
      delay(5000);
    }
    else
    { 
      //get ssid and password from portal
      String ssid = wm.getWiFiSSID();
      ssid.toCharArray(ssidString, sizeof(ssidString));
      String pw = wm.getWiFiPass();
      pw.toCharArray(pwString, sizeof(pwString));
      
      //save ssid and password to preferences
      preferences.begin("credentials", false);
        preferences.putString("ssid", ssidString); 
        preferences.putString("password", pwString);
      preferences.end();
    }
  }
}

tablatronix, 2025-02-01:
If you remove getwificredentials?

GrantSando, 2025-02-01:
Thanks for your reply. I have now simplified the code and uninstalled all additional libraries apart from WiFiManager.h and still get the same result. (starts portal with 3.0.7, crashes chip with 3.1.0). I should add this is Arduino IDE 2.3.4

Are there some other files I should delete before testing?

Code now is, 

#include <WiFiManager.h>
WiFiManager wm;   

void setup() {
    Serial.begin(9600);
    delay(1500); 

    WiFi.mode(WIFI_STA);
    wm.startConfigPortal("MY_WIFI", "PASS1234");

}

void loop() {
  // put your main code here, to run repeatedly:

}

GrantSando, 2025-02-01:
chip is: ESP32-S3-WROOM-1-N8
partition scheme is: 8mb with spiffs (or 4mb with spiffs - same result)

maxzerker, 2025-02-06:
Same issues here, ESP32

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401962e2  PS      : 0x00060230  A0      : 0x800e6088  A1      : 0x3ffb2090  
A2      : 0x00000200  A3      : 0x3f4003d8  A4      : 0x3f401060  A5      : 0x00000001  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x800f4410  A9      : 0x3ffb20c0  
A10     : 0x00000002  A11     : 0x3f408898  A12     : 0x0000001c  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000022d  LBEG    : 0x4008be7d  LEND    : 0x4008be8d  LCOUNT  : 0xffffffff  

Backtrace: 0x401962df:0x3ffb2090 0x400e6085:0x3ffb20b0 0x400d5263:0x3ffb2120 0x400d5367:0x3ffb2200 0x400f9cbb:0x3ffb2270 0x400910e2:0x3ffb2290

PC: 0x401962e2: WiFiManager::_begin() at C:\Users\maxze\SynologyDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:249
EXCVADDR: 0x0000022d

Decoding stack results
0x401962df: WiFiManager::_begin() at C:\Users\maxze\SynologyDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:249
0x400e6085:  is in WiFiManager::startConfigPortal(char const*, char const*) (C:\Users\maxze\SynologyDrive\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:698).
0x400d5263: wifiManagerSetup() at C:\Users\maxze\SynologyDrive\Documents\Arduino\myprj\_ESP32\ANPR-Led-Matrix-Display\ANPR-Led-Matrix-Display.ino:154
0x400d5367: setup() at C:\Users\maxze\SynologyDrive\Documents\Arduino\myprj\_ESP32\ANPR-Led-Matrix-Display\ANPR-Led-Matrix-Display.ino:392
0x400f9cbb: loopTask(void*) at C:\Users\maxze\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\cores\esp32\main.cpp:59
0x400910e2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa\port.c:162

tablatronix, 2025-02-09:
Works for me, super example

```
*wm:[2] [SYS] WM version:  v2.0.17 D:4
*wm:[2] [SYS] Arduino version:  3.1.1
*wm:[2] [SYS] ESP SDK version:  5.3.2.250120
*wm:[2] [SYS] Free heap:        347712
*wm:[2] [SYS] Chip ID: 3245692124
*wm:[2] [SYS] Chip Model: ESP32-S3
*wm:[2] [SYS] Chip Cores: 2
*wm:[2] [SYS] Chip Rev: 1
```

``````
[  4003][V][AP.cpp:108] _onApEvent(): AP Started
[  4007][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 130 - AP_START
[  4015][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 130 - AP_START
*wm:[2] AP IP address: 192.168.4.1
*wm:[3] [CB] _apcallback calling 
[CALLBACK] configModeCallback fired
*wm:[4] setupConfigPortal 
*wm:[2] Starting Web Portal 
[  4503][V][NetworkServer.h:41] NetworkServer(): NetworkServer::NetworkServer(port=80, ...)
[  4512][V][WebServer.cpp:49] WebServer(): WebServer::Webserver(port=80)
*wm:[3] [CB] _webservercallback calling 
*wm:[3] HTTP server started 
*wm:[4] setupDNSD 
*wm:[4] dns server started with ip:  192.168.4.1
*wm:[3] Config Portal Running, blocking, waiting for clients... 
*wm:[3] Portal Timeout In 90 seconds
*wm:[3] Portal Timeout In 63 seconds
```

GrantSando, 2025-02-09:
OK thanks, just tested that and it worked for me also.

I noticed WiFi.mode(WIFI_STA) was removed from the super example. When I remove this from my examples above they also work without issue.

Seems that with core 3.1.0 and higher this line shouldn't be called. 

Thanks for your help and the work you've done with this library.

tablatronix, 2025-02-09:
Ah good point, could be a race condition
But that also is working for me without issue