tzapu/WiFiManager #1803: Trying to reduce the size.....

bentwonk, 2025-02-19:
OK I am a ESP32 novice, I need what this library does, allow the non coders too easily connect an ESP32 to their home Wifi, but damn is this thing large, sure I want to get the local Wifi details , but then I want space to run my own code, fun things like Rest API, MQTT, logic, sensor polling and cleanup, OTA via website. So I took hacked away big chucks of the code I do not need, and yet it still takes 53% of my minimal Spiffs config, seems fat to me for something which only needs to get the users Wifi details, how do I reduce it further?

tablatronix, 2025-02-19:
There is a min branch I made to test this out, but theres not much you can do, the esp library has also grown.
Like you say I cant get it under a certain size. Even removing all strings. Try a different partition scheme, ota and spiffs is alot to ask for woth other libs, you using 4M?

tablatronix, 2025-02-19:
I havent tried removing wifi ap scanning, have to manual enter wifi then..

bentwonk, 2025-02-22:
Thank you,  will see if I can make any progress with platformio.

Grey-Lancaster, 2025-05-16:
Hey, I am just a lurker, but I use this library wilt platformio.  It does not go into spiffs ???
https://github.com/Grey-Lancaster/IncuBuddy