tzapu/WiFiManager #1805: setCustomBodyHeader not working?

MYCAMEL222, 2025-03-02:
I am new to WM, but have somewhat figured my way around it. Great Library!

using tag 2.0.17 (latest).

I am trying to use setCustomBodyHeader. setCustomBodyFooter is not working either. 

I get this error....Compilation error: 'class WiFiManager' has no member named 'setCustomBodyHeader'; did you mean 'setCustomHeadElement'?

setCustomHeadElement works fine.......

  wm.setCustomHeadElement("<style>div[class='msg'],form[action='/wifi'],form[action='/0wifi'],form[action='/info']:nth-of-type(2){display:none}</style>");  

What am I missing?

dmadison, 2025-03-04:
Those functions are new (8e4c612fa3af0e3eb9d98b7a6da4acec0a37744a) and there has not been a release since they were added. If you want to use them you need to download the version of the library on master instead of using the latest release.

MYCAMEL222, 2025-03-05:
That (of course) fixed it.

Thanks!...

closed 2025-03-05 by MYCAMEL222