tzapu/WiFiManager #1808: In config portal, while sending Configure WiFi page, can I send it without scanning available connection?

y-runwal, 2025-03-12:
Hello,
This library is great, many thanks to the development community !!!

Q. In config portal, while sending Configure WiFi page, can I send it without scanning available connection? Like just give fields to enter wifi ssid and pass.

Thank you in advance!!!

y-runwal, 2025-03-12:
I am asking this because ESP32 says that config page has been sent. But on chrome there is nothing displayed.

timr49, 2025-04-17:
> Q. In config portal, while sending Configure WiFi page, can I send it without scanning available connection? Like just give fields to enter wifi ssid and pass.

Hi @y-runwal, from looking at the code if seems that the URL path /wifi gives the wifi configuration page with scanning, but the URL path /0wifi gives the wifi configuration page **without** scanning. Specifically, when the WiFiManager is running and you have connected to its WiFi Access Point, type the address http://192.168.4.1/0wifi into your browser (the 0 is a zero not a letter o).

Does that help?

y-runwal, 2025-04-28:
Hi @timr49, thank you for reply. I ended up removing wifi page and added custom parameters to get these from user itself. Still thank you for sharing this info.

closed 2025-04-28 by y-runwal

tablatronix, 2025-04-30:
Change the menu to use the no scan page