tzapu/WiFiManager #1817: Wifi Manager - Station Issue

Sul-M, 2025-04-25:
Hi - managed to install WifiManager and connect AP without issue. I can also connect to Wifi but when I type the IP address into a browser it does not resolve.
I have tried this via router and mobile hotspot with the same issue.

IP range, subnet etc looks within range. If I run the Wifi example where I code in the SSID/Password it works fine. Makes me think there is an issue with WifiManager?

Appreciate if anyone could help me sort (Also FYI - I am using the Wifi Manager onDemand/OnDemandConfigPortal)
Thanks in advance,

Sul-M, 2025-04-27:
Hi Tim,
I'm very new to this, and think I worked out the initial issue having looked at all the code.
In summary I have used wifimanager to create a AP and then Station for my ESP32, I created a webserver with code to pull the wifimanager saved wifi network too which is great.
The issue I have now is linking wifimanager to connect wifi on a SD card for read/write... I have managed to set it up using a wifi.softAP() but want it to connect via the wifimanager code.

In the supporting docs it lists wifi.begin() to be changed with code from wifimanager but unclear on what to change on this SD sketch that has wifi.softAP().

Hope I have explained my issue clearly enough?
Thanks

timr49, 2025-04-28:
> Hi Tim, I'm very new to this, and think I worked out the initial issue having looked at all the code.  ...
No worries, we have  all been there!  I will keep asking for clarifications, it's just part of the process.

>In summary I have used wifimanager to create a AP and then Station for my ESP32, ...
If you have got the ESP32 to connect to your usual wireless network after using SSID & password details entered with WiFiManager in AP mode, then the only remaining thing to do with WiFiManager is to get it to reconnect to the wifi every time your sketch runs and when it loses a connection.

>I created a webserver with code to pull the wifimanager saved wifi network too which is great.
Typically, WiFiManager handles the wifi  connection before a web server is started, so I am not sure why the web server needs to pull anything from WiFiManager (unless you are using WiFiManager to collect configuration information in addition to the SSID & password).

>The issue I have now is linking wifimanager to connect wifi on a SD card for read/write... I have managed to set it up using a wifi.softAP() but want it to connect via the wifimanager code.
I can't see a relationship between wifi and an SD card so please explain what it is you want to do with the SD card. For example, do you mean that you want a web server to provide a user with a web browser with access to the files on an SD card? In that case, I suggest looking at "FSBrowser" in the ESP32 WebServer examples.

> In the supporting docs it lists wifi.begin() to be changed with code from wifimanager but unclear on what to change on this SD sketch that has wifi.softAP().
Is the "SD sketch" one of the examples that comes with a library? If so, which one? If not, can you post it?

Regards,
-Tim.

Sul-M, 2025-04-28:
Once again thanks Tim, really appreciate you taking time to respond!
Indeed I am trying to access the SD card via connection made with Wifimanager.

I was following the below from YouTube but unclear on what to change/add to establish the connection via Wifimanager.
https://www.youtube.com/watch?v=zoYMU1tA3nI&list=PLQBtkGz5Mr6rgEeMTs1LB6UU2egOyXLKz&index=6&t=381s

I was also looking at this but seems a little more complex to me.
https://github.com/G6EJD/ESP32-8266-FIle-Upload?tab=readme-ov-file

Regards,
Sul

timr49, 2025-04-28:
> ...Indeed I am trying to access the SD card via connection made with Wifimanager.

I recommend that you start with the "FSBrowser" example in the ESP32 WebServer examples using a hardcoded SSID and password. When you have that working, add WiFiManager to replace the hardcoded wifi credentials, choosing how to do that from the most appropriate of the WiFiManager examples.

Regards,
-Tim.

Sul-M, 2025-04-28:
Ok I will try what you suggested...where do I get the FSBrower example from, in the ESP32 WebServer I see two examples, ESP32webserver1 and ESP32webserver2? 
Thanks,
Sul

Sul-M, 2025-04-29:
Thanks once again Tim.
I will take a look at Randomnerdtutorials, I have seen other guides from them that were great.
Noted on the random YouTube videos.
Rgds,
Sul